bambulab/BambuStudio #6441: Not all objects are sliced

adam-puleo, 2025-04-05:
### Bambu Studio Version

2.0.1.50

### Where is the application from?

Bambu Lab Official website

### OS version

macOS 15.4

### Additional system information

CPU: Apple M1 Pro

### Printer

Bambu Lab X1-Carbon

### How to reproduce

1. In the "Prepare" window notice that in the upper right for the "Bottom" and "Middle" plates there's a big blue "B".
2. In the "Preview" window the "B" is missing in the "Bottom" and "Middle" plates.

### Actual results

The B is missing in the Bottom and Middle plates.

### Expected results

B should be present.

### Project file & Debug log uploads

[log.zip](https://github.com/user-attachments/files/19617865/log.zip)
[Playing Card Holder.3mf.zip](https://github.com/user-attachments/files/19617866/Playing.Card.Holder.3mf.zip)

### Checklist of files to include

- [x] Log file
- [x] Project file

Haidiye00, 2025-04-07:
[image]
[image]
Hello, I suggest adjusting the height of the letter "B" in the z-direction

adam-puleo, 2025-04-08:
@Haidiye00, interesting, that seems to work. It just will not be flush with the brown plate though.

Any idea why just the B is not sliced?

Thanks

adam-puleo, 2025-04-08:
@zhimin-zeng-bambulab, I changed B's global coordinates' Z to 3.1mm. When I slice the model the B only shows up in the 21 layer (4.2mm). The B object is 2mm thick so shouldn't it show up in 10 layers?

Thank you