bambulab/BambuStudio #6443: Frequent crashes when connected to dual monitors (2 x 2k 32")

GuyH77, 2025-04-05:
### Bambu Studio Version

2.0.1.50

### Where is the application from?

Bambu Lab Official website

### OS version

Win 11

### Additional system information

OS Name	Microsoft Windows 11 Enterprise
Processor	13th Gen Intel(R) Core(TM) i7-13850HX, 2100 Mhz, 20 Core(s), 28 Logical Processor(s)
System Model	HP ZBook Fury 16 G10 Mobile Workstation PC

### Printer

Bambu Lab H2D

### How to reproduce

I am very happy to help or supply logs but this was a strange one. Until this release I had no problem. Having received my H2D and upgraded I experienced crash after crash. The software is unusable. I have 2 32" 2k monitors attached to a HP USB dock. It was only once I disconnected that Bambu Studio could be used again. I would be very happy to help find out the cause of this issue.

### Actual results

Crash when I sliced, saved or sync'd

### Expected results

No crash

### Project file & Debug log uploads

It's not project related

### Checklist of files to include

- [ ] Log file
- [ ] Project file

BoBoYu-1, 2025-04-07:
Thank you for your feedback. Could you please provide the log file and DMP file?

GuyH77, 2025-04-07:
[log.zip](https://github.com/user-attachments/files/19627611/log.zip)

@BoBoYu-1 I've attached the logs of recent crashes. I believe it's likely to do with graphics but can't be sure. Some of my system settings and settings in BS seem to have made it a little more stable but I still get crashes when slicing, changing settings and saving.

[image]

[image]

[image]

Driver info.

[image]

GuyH77, 2025-04-07:
Another crash. Logs and 3mf attached. 

[BambuStudioCrashLog.zip](https://github.com/user-attachments/files/19632330/BambuStudioCrashLog.zip)

Haidiye00, 2025-04-07:
[image]
Hello, turn off this option and see the result

GuyH77, 2025-04-07:
> [image] Hello, turn off this option and see the result

Hello, I've tried both on and off. I still have the crashes.

GuyH77, 2025-04-11:
An update. I uninstalled the nVidia usb driver that is installed with the graphics driver. I've also reverted to an older driver as well. So far, no crashes. My HP zbook is connected to the monitors through the HP dock which uses USB-C. I wonder if this is the thing that doesn't play nicely with Bambu Studio. My other apps have been fine so I think you should investigate this. Thanks.