bambulab/BambuStudio #6444: AMS Sync Loads ABS

Savemeeeee, 2025-04-06:
### Bambu Studio Version

2.0.1.50

### Where is the application from?

Bambu Lab Official website

### OS version

Windows 11

### Printer

Bambu A1

### How to reproduce

Load into Prepare it already shows ABS as a filament
Re-syncing filament from AMS Lite adds everything currently on my AMS lite plus ABS and another filament. 

### Actual results

When loading into the "Prepare" Tab this shows

[image]

After Re-syncing filaments from AMS Lite it shows ABS, however I just tried to recreate it after the fourth or fifth time of it happening and it worked like it's supposed to. 

### Expected results

[image]

### Project file & Debug log uploads

[log.zip](https://github.com/user-attachments/files/19619429/log.zip)
[Bug.zip](https://github.com/user-attachments/files/19619428/Bug.zip)

### Checklist of files to include

- [x] Log file
- [x] Project file

Savemeeeee, 2025-04-07:
> Hello, could you please take a screenshot of the consumables on the device
> page? I didn't quite understand what you meant~

It won’t be until at least Thursday. But I’ll try to remember

Savemeeeee, 2025-04-10:
[image: image.png]
That's what shows up when I load into bambu studio. I do not have ABS
loaded onto my AMS Lite nor have I ever.

After clicking on the re-sync page, this is what shows up
[image: image.png]

I'm not sure if this what you're asking for
[image: image.png]

Haidiye00, 2025-04-10:
Hello, your picture was not uploaded successfully. Could you please upload it again? Thank you

Haidiye00, 2025-04-16:
Hello, please use the latest version, thank you.
https://github.com/bambulab/BambuStudio/releases/tag/v02.00.02.57

Savemeeeee, 2025-04-16:
I didn’t have this issue until I updated to the latest version

closed 2025-04-16 by Haidiye00