bambulab/BambuStudio #6445: Slicing the same model more than once causes crash

Nooka6845, 2025-04-06:
### Bambu Studio Version

2.0.1.50

### Where is the application from?

Bambu Lab Official website

### OS version

Windows 11 v24h2 26100.3476

### Printer

Any

### How to reproduce

Open Model to slicer.
Slice model
Back to Prepare
Make modification
Slice model again
program crashes

### Actual results

have to completely restart program and slicing

### Expected results

no crashes

### Project file & Debug log uploads

[crash_Sun_Apr_06_01_24_14_0_done.log](https://github.com/user-attachments/files/19619974/crash_Sun_Apr_06_01_24_14_0_done.log)

### Checklist of files to include

- [ ] Log file
- [ ] Project file

Nooka6845, 2025-04-06:
Update:
It got WORSE....
I tried just slicing on 1 monitor. When having multiple models on 1 plate, and slicing it will, slice successfully, but you HAVE to be fast to be able to send it to a printer. Otherwise after maybe 10-20 seconds, program closes and you have to start over.

BoBoYu-1, 2025-04-07:
Thank you for your feedback. Please provide the log file and DMP file, and also provide the makeworld model connection so that we can reproduce the problem and solve it.
The steps to obtain the logs are as follows：https://wiki.bambulab.com/en/bambu-studio/troubleshoot/crash-freeze-issue

BoBoYu-1, 2025-04-07:
Thank you for your feedback. Please provide the DMP file
The steps to obtain the logs are as follows：https://wiki.bambulab.com/en/bambu-studio/troubleshoot/crash-freeze-issue

QingZhangBambu, 2025-05-23:
@Nooka6845 could you provid the 3mf file? i will try to reproduce your case

QingZhangBambu, 2025-06-03:
OK :)