bambulab/BambuStudio #6451: v2.0.1.50 crash during slicing

Zratt, 2025-04-06:
### Bambu Studio Version

v2.0.1.50

### Where is the application from?

Bambu Lab Official website

### OS version

Win11

### Additional system information

CPU: Intel 10900kf
RAM: 32GB
GPU: RTX 3080

### Printer

Bambu P1S

### How to reproduce

Try to slice any model and the app crashes around 80% sliced

### Actual results

Bambu Studio crashes

### Expected results

The model should be sliced and ready to print.

### Project file & Debug log uploads

[bambu-studio.exe.2704.zip](https://github.com/user-attachments/files/19622233/bambu-studio.exe.2704.zip)

### Checklist of files to include

- [x] Log file
- [ ] Project file

GuyH77, 2025-04-06:
You're not alone. Hoping Bambu can identify the issue and fix. I'm seeing Windows Defender ramp up more but not sure if it's just coincidence.

pgifford, 2025-04-07:
How about adding a project file because I can slice all day every day with 2.0.1.50 on Windows 10.

BoBoYu-1, 2025-04-07:
Thank you for your feedback. Could you please provide the 3mf file?

GuyH77, 2025-04-07:
> Thank you for your feedback. Could you please provide the 3mf file?

I assume this is for the OP? I've started my own bug and shared logs

QingZhangBambu, 2025-05-23:
@Zratt Apologies for replying late. Could you please try this method? It has worked for most people.
https://wiki.bambulab.com/en/bambu-studio/troubleshoot/win-crash-when-slicing

closed 2025-07-21 by QingZhangBambu