bambulab/BambuStudio #6453: Cannot use Capital letter "D" when trying to add text to bambu studio. What gives?!

jvalal, 2025-04-06:
### Bambu Studio Version

2.0.1.50

### Where is the application from?

Bambu Lab Official website

### OS version

MAC OS Sequoia Version 15.3.2 (24D81)

### Additional system information

When holding down the shift button to type a capital "D" in the text field, you cannot type the capital letter D. COME ON GUYS!!

### Printer

A1

### How to reproduce

When holding down the shift button to type a capital "D" in the text field, you cannot type the capital letter D. COME ON GUYS!!

### Actual results

Nothing happens, 

### Expected results

the letter "D" to be typed!

### Project file & Debug log uploads

N/A

### Checklist of files to include

- [ ] Log file
- [ ] Project file

pgifford, 2025-04-07:
Works for me.

> When holding down the shift button to type a capital "D" in the text field, you cannot type the capital letter D.

No, _you_ cannot type the capital letter D.  I can.  Did you try pasting a capital D?  Could you type a lowercase d?  You didn't upload a project file or include screenshots so the engineers have to guess at your settings?

Close this issue, not enough information.

[image]

pgifford, 2025-04-07:
Also close because it's a duplicate of the better-written #6430 and there's a workaround in [a comment](https://github.com/bambulab/BambuStudio/issues/6430#issuecomment-2781958280).

jvalal, 2025-04-07:
Don't dictate orders a-hole! It's not helpful one bit and u know it.
Pasting works
Lower case works
SHIFT D doesn't work

Haidiye00, 2025-04-07:
Hello, this is a bug. We will fix it in the next version.
At present, you can input it elsewhere and then copy it in. We apologize for any inconvenience caused

closed 2025-04-16 by Haidiye00