bambulab/BambuStudio #6454: 3mf generated by an old version of Bambu Studio, loading geometry data only.

LordIrishDave, 2025-04-07:
### Bambu Studio Version

02.00.01.50

### Where is the application from?

Bambu Lab Official website

### OS version

Mac Sequoia 15.4

### Printer

X1C

### How to reproduce

Just launch BS and try to import any drawing file exported from Fusion 360 in a 3mf format.  

### Actual results

[image]

### Expected results

No error codes or error messages.  This just recently happened when I installed the last few updates released in the last 10 days

### Project file & Debug log uploads

[Kim's Couch Legs v1.3mf.zip](https://github.com/user-attachments/files/19624320/Kim.s.Couch.Legs.v1.3mf.zip)

### Checklist of files to include

- [ ] Log file
- [x] Project file

Haidiye00, 2025-04-07:
[image]
Hello, our internal version has completed the optimization of pop ups. Thank you for your discovery~

LordIrishDave, 2025-04-07:
What does that mean for me??

Is the problem solved?  Was this an issue on your side or my side??

Do I need to do anything??

KopiTubruk, 2025-04-07:
Hello,

I have the same problem.

I am trying to import a 3mf file to OnShape. The error message already mentioned appears. In addition, Bambu Studio complains about wrong scaling and then about different heights in the individual objects in the file. BambuStudio also does not recognize all parts, parts of the 3D drawing are missing.  The 3mf is fine, 3Dviewer.net shows it correctly and recognizes all parts.

Thx,
Kopi

Haidiye00, 2025-04-08:
Hello, the pop-up message is incorrect. We will fix it in the next version~

closed 2025-04-16 by Haidiye00