bambulab/BambuStudio #6456: Bambu Lab crashes after slicing in preview mode

b4after-pl, 2025-04-07:
### Bambu Studio Version

2.0.1.50

### Where is the application from?

Bambu Lab Official website

### OS version

Windows 11

### Additional system information

Microsoft Windows 11 Pro, Wersja 10.0.26100
DirectX Runtime: DirectX 12
Graphics driver: NVIDIA RTX Enterprise - 572.83 - Tue Mar 18, 2025
Processor: Intel(R) Core(TM) i9-10940X CPU @ 3.30GHz

Graphics card: Nvidia Quadro RTX 4000
Direct3D: 12_1
CUDA: 2304
Ram on card: 8192 MB GDDR6

### Printer

A1, P1S, X1C

### How to reproduce

1. Place one complex model or many copies of small one on build plate (in my case it was 196 copies of borad game cubes)
2. Click slice
3. Bambu Studio will switch to preview mode and after slicing is done it crash
4. To avoid crash I need to switch back to Prepare mode (or any other than preview) and wait to slice will be done.
5. Then I can send it to printer, but if I switch to Preview Bambu Studio will crash.

### Actual results

Crashes when preview sliced complex model

### Expected results

Works correct. Not crashing.

### Project file & Debug log uploads

[log.zip](https://github.com/user-attachments/files/19628532/log.zip)

[A1 pojemniki.zip](https://github.com/user-attachments/files/19628529/A1.pojemniki.zip)

### Checklist of files to include

- [x] Log file
- [x] Project file

b4after-pl, 2025-05-13:
Any one can replicate issue or have some suggestions how to fix?

QingZhangBambu, 2025-05-23:
@b4after-pl Apologies for replying late. Could you please try this method? It has worked for most people.
https://wiki.bambulab.com/en/bambu-studio/troubleshoot/win-crash-when-slicing