bambulab/BambuStudio #6461: Prepare tab freez after first slicing

hbock2, 2025-04-07:
### Bambu Studio Version

2.0.1.50

### Where is the application from?

Bambu Lab Official website

### OS version

windows 11

### Additional system information

CPU: Intel i7 13700K
GPU: Asus 4070TI
Motherboard: Asus PRIME Z790-A WIFI

### Printer

Bambu P1S

### How to reproduce

1. Load any model
2. Slice it
3. Go back to prepare tab

### Actual results

View of model freeze, but when you try to move it "blind" and slice again it moves what you can observe in preview tab. Restart app helps but only till next slice. 
Reinstall GPU drivers helps for couple of slicing. 

### Expected results

Model should move. 

### Project file & Debug log uploads

[log.zip](https://github.com/user-attachments/files/19636282/log.zip)

### Checklist of files to include

- [x] Log file
- [ ] Project file

Haidiye00, 2025-04-08:
Hello, can you download ScreenToGif software and record a GIF of the operation video and upload it? Thank you
[Release ScreenToGif 2.41.1 · NickeManarin/ScreenToGif](https://github.com/NickeManarin/ScreenToGif/releases/tag/2.41.1)

hbock2, 2025-04-08:
[image]

Haidiye00, 2025-04-09:
Hello, looking at your animation, this horse is still on the plate, but there is no content in the slice? Do you mean this? Can you upload these two 3mf files? Thank you

hbock2, 2025-04-09:
The problem is that after slicing, when I go back to the Prepare tab, I can't rotate the bed—it won't move at all. I have to restart the program to make it work again, but it only works until the next slicing.

hbock2, 2025-04-09:
Ok, I think I found solution. After closing Asus app "GPU tweak III" everything works great.

Haidiye00, 2025-04-10:
Okay, thank you for your feedback~

closed 2025-04-16 by Haidiye00