bambulab/BambuStudio #6462: Support Expansion not expanding interface

Festivejelly, 2025-04-07:
### Bambu Studio Version

2.0.1.50

### Where is the application from?

Bambu Lab github releases

### OS version

Windows 10

### Printer

H2D, X1C

### How to reproduce

Add a part that has a circle hole that needs support.
Under support settings set support expansion to 0.5mm
Observe layers

### Actual results

The final support interface layer is inset

[image]

### Expected results

The final support interface layer should like up with the layer below

[image]

### Project file & Debug log uploads

Proprietary file, cant share.

### Checklist of files to include

- [ ] Log file
- [ ] Project file

NothFie, 2025-04-09:
@Festivejelly Thanks for your report. This is a BUG and we are fixing it. For now if this issue matters much, you could try Grid Support and it might be able to cover the hole and avoid the "Flying lines".

Festivejelly, 2025-08-30:
Its still not really fixed esp when supporting round holes