bambulab/BambuStudio #6463: Add "Keep Existing Printer" and "Keep Existing Filaments" Option

glizzyglizz69, 2025-04-07:
My problem is that, when using a .3mf file, I am forced to always change my printer and filaments settings. 

This is common practice, however, I believe nearly everyone using a .3mf file is changing their settings. This because it is highly unlikely that the creator will have the identical printer, nozzle and filament settings. Furthermore, why does the option to choose your printer profile before downloading a model from Maker World exist? I have not had it download models with the specific printer I am choosing? 

I understand the features of a .3mf file serve a purpose. They are much more robust than an .stl. However, the Slicer (Bambu Studio) should offer the option to the user to "keep existing printer" and "keep existing filament". This type of customization and freedom is what we see for print settings when being compared to the base profile. It's wonderful to have that little orange reset button, but why not give us the option for the printer and filament settings? I know this is probably a niche issue, but I really think people would start to use it by default. It should be the default. Like I said, giving us the option to do this would save everyone time, and it just makes things that much smoother. Please add this option to Bambu Studio.

fatty0matty, 2026-11-15:
100% there should be an option to keep your existing printer. There is never a logical reason to switch my printer to a printer I don’t own. The fact that is even an option shows somebody without a brain designed this software. 
And then Bambu’s solution, that we should all manually change back our printer on every single file we open from MakerWorld - instead of them just changing it so that we can keep our printer?? It just shows just a complete lack of basic reasoning.

Mujin-code, 2026-11-16:
你好，我们会考虑你的优化建议。
在你导入文件之后直接点击通过按钮，软件会将你的打印机信息同步到项目中，项目耗材也可以直接一键同步，你可以尝试一下同步功能

[image]