bambulab/BambuStudio #6466: 安装两个版本的时候每次打开都会显示安装网络插件

ceca00, 2025-04-08:
### Bambu Studio Version

2.0.1.50

### Where is the application from?

Bambu Lab github releases

### OS version

windows10

### Printer

p1s

### How to reproduce

因为切片逻辑的原因，我电脑不得不安装两个版本软件，最新版本的软件是通过exe安装包安装的，路径为默认路径，还有一个1.9.2.57版本是通过下载zip解压使用的，当我每次打开软件时，不管哪个版本，都会显示正在装网络插件
[image]

而且当我打开了一个版本的软件后，再打开另一个版本时会显示安装插件失败

[image]

### Actual results

显示下载插件

### Expected results

直接显示主页

### Project file & Debug log uploads

nofile

### Checklist of files to include

- [ ] Log file
- [ ] Project file

BoBoYu-1, 2025-04-08:
抱歉，1.9和2.0的插件版本不同，为了不影响功能使用所以每次运行都要重新下载插件；暂时没办法同时存在两个不同版本的插件

BoBoYu-BamBu, 2025-04-28:
@ceca00  Hello, have your questions been resolved?