bambulab/BambuStudio #6470: Can´t change Colours of the filament in the ams

Michadennerl, 2025-04-08:
### Bambu Studio Version

2.01.50

### Where is the application from?

Bambu Lab Official website

### OS version

WIN 11

### Printer

P1S Firmware 1.07.00.00

### How to reproduce

I want to change the colours and the settings of the filaments in the ams. 
When I change the colours and the type in the app, there is "fail".
When I change the settings in the studio, it changes for 5 seconds, and the it goes back to the colour before.
I have a 6 colour print and its very complicated to print with this setup.
The Problem is since two weeks. Before everything was good.

### Actual results

nothing

### Expected results

I can change the colour like before.

### Project file & Debug log uploads

### Checklist of files to include

- [ ] Log file
- [ ] Project file

Haidiye00, 2025-04-08:
Hello, can you download ScreenToGif software and record a GIF of the operation video and upload it? Thank you
[Release ScreenToGif 2.41.1 · NickeManarin/ScreenToGif](https://github.com/NickeManarin/ScreenToGif/releases/tag/2.41.1)

Michadennerl, 2025-04-08:
[image: Bambu studio.gif]
Hello, here are the two recordings.
One ist the bambu studio, one is the app.
You have to wait a few minutes at the studio vid, you can see the problem
at the end.

My normal operation is, that I put my whole filaments in both AMS, then I
change the colours in the app.
When I am at the Desktop PC, I syncronise the Studio with the app and start
the printing.
But since two weeks, I can´t change the colour.

Haidiye00, 2025-04-09:
Hello, your upload failed. Can you please try uploading again? The upload was successful, and you can also see the GIF animation

Michadennerl, 2025-04-09:
https://github.com/user-attachments/assets/e6910394-f88a-4fb5-a8c8-66fa1bc63186

[image]

northman66, 2025-04-09:
I am hitting this exact same issue
try to set it, it sets for a little less than 10 seconds, then goes back to unset

Redbeard74, 2025-04-10:
Having the exact same issue. Just put in a support ticket.

Notadouche, 2025-04-10:
Add me to this list. Being a real pain. I also have 2 AMS with my P1S. I first thought it was a problem with one track so I started unloading and reloading other ones, now some aren't even recognized anymore. Down to 5 colors out of 8.... We need a fix :)

gaolegao-lx, 2025-04-11:
**Please upload log through Handy APP, and give us your ticket number**
***
***Upload log files through Bambu Handy***
     Log files from the Bambu printer can be uploaded via Bambu Handy over the network, with an upload speed of around 150 kbps under optimal network conditions.
     When uploading logs, you have the option to select a specific time frame, allowing you to reduce the volume of logs transmitted and shorten the upload duration.
     To upload files, go to **Me** → **Support Tickets** → Choose the ticket for log upload and proceed with the operation.
     [image]

Notadouche, 2025-04-11:
Hello!

I have uploaded the logs from my SD card and now the Handy App. Here is my ticket: US250410555001

Thanks!

odin6796, 2025-04-11:
I have the same problem....

Michadennerl, 2025-04-11:
I couldn´t find my actual suppport ticket in the studio. I made new one and
wrote the issue #6470 in the text.
The number of the Studio ticket is US250411195001

Skye-Bambulab, 2026-11-16:
Hello! Currently, we have not been able to reproduce this issue in the latest version.

Therefore, please upgrade your [BambuStudio/BambuHandy] to the latest release [BambuStudio: 02.09.01.70 / BambuHandy: 4.4.2 for Android, 4.4.3 for iOS]. If the issue still persists, please submit a support ticket through BambuHandy or the official website and upload new logs for our analysis. You can refer to the wiki(https://wiki.bambulab.com/en/x1/troubleshooting/update-log-to-a-ticket) for instructions.

Thank you for your understanding and support!