bambulab/BambuStudio #6473: Wrong color used since last major update

jup2000, 2025-04-08:
### Bambu Studio Version

2.0.1.50

### Where is the application from?

Bambu Lab Official website

### OS version

Win 11

### Additional system information

i7 96gb ram nvidia 3070

### Printer

X1C

### How to reproduce

Select device, select model, select AMS color slot 3 (or whatever)

### Actual results

Starts printing with the WRONG color.  I select slot 3, it decides it wants to print using slot 1.  This has NEVER happened until the last major update with the new UI.  It is extremely frustrating as I can no longer trust sending prints via BBStudio.

### Expected results

I expect it to use the damn slot I told it too.

### Project file & Debug log uploads

Proprietary stl files, can't publish.  Has nothing to do with project.  Does this with new or old projects.

### Checklist of files to include

- [ ] Log file
- [ ] Project file

Haidiye00, 2025-06-03:
Hello, please use the latest version of the software. Thank you~
https://github.com/bambulab/BambuStudio/releases/tag/v02.01.00.59

Haidiye00, 2025-07-17:
https://github.com/bambulab/BambuStudio/releases
Hello, please use the latest beta version. Thank you~
If you have any further questions, we will analyze again

closed 2025-07-17 by Haidiye00