bambulab/BambuStudio #6474: Can I have two computers running Bambu Studio software connected to  an X1 Carbon? If so my second computer is not connecting.

BambuBongo, 2025-04-08:
### Bambu Studio Version

2.0.0.95

### Where is the application from?

Bambu Lab Official website

### OS version

Windows 11

### Additional system information

Device name	DESKTOP-FBMPOFK
Processor	11th Gen Intel(R) Core(TM) i5-1135G7 @ 2.40GHz   2.42 GHz
Installed RAM	8.00 GB (7.74 GB usable)
Device ID	9B64EC4C-649E-447D-B5C4-CAC3CE4C03BA
Product ID	00356-07261-79936-AAOEM
System type	64-bit operating system, x64-based processor
Pen and touch	Touch support with 10 touch points
Edition	Windows 11 Home
Version	24H2
Installed on	‎1/‎31/‎2025
OS build	26100.3624
Experience	Windows Feature Experience Pack 1000.26100.66.0

### Printer

BambuLab X1 Carbon

### How to reproduce

Can't find my device when adding printer
Tried adding with Access code and I.P address from printer . Unable to connect.
Cannot locate Pin Code in Account information on printer.

### Actual results

Unable to connect to printer.

### Expected results

Should be able to make connection to the printer.

### Project file & Debug log uploads

Not running a project. Just trying to make connection to the printer from a second computer running Bambu Studio.

### Checklist of files to include

- [ ] Log file
- [ ] Project file

BoBoYu-1, 2025-04-10:
Machines in non-local area network mode cannot be bound to multiple accounts. If you want to bind different accounts to the same machine, you can enable the local area network mode on the printer screen.

BambuBongo, 2025-04-10:
Thanks for the feedback. This helps a lot
Take care.

BoBoYu-1, 2025-04-11:
Your problem has been resolved. Can we close this issue now?

BoBoYu-BamBu, 2025-04-28:
@BambuBongo Your problem has been resolved. Can we close this issue now?

closed 2025-04-29 by BoBoYu-BamBu