bambulab/BambuStudio #6475: Bambu Studio crashes during mesh boolean

seanbo, 2025-04-09:
### Bambu Studio Version

2.0.1.50

### Where is the application from?

Bambu Lab github releases

### OS version

Windows 11

### Additional system information

CPU: Ryzen 5
RAM: 64GB
GPU: Nvidia 4090ti

### Printer

Bambu Lab A1

### How to reproduce

1. Open RepRack.3mf
2. Go to plate: Dowel Hole Plugs
3. Select part and click Mesh Boolean button on toolbar
4. Select "Outer Wall" for Part 1
5. Select "Inner Ring" for Part 2
6. Using the Union option, click the Union Button
7. *This is where the crash occurs*

**also**

1. Open RepRack.3mf
2. Go to plate: Dowel Hole Plugs
3. Select part and click Mesh Boolean button on toolbar
4. Select "Outer Wall" for Part 1
5. Select "Generic-Cylinder" for Part 2
6. Select "Delete input" or not, both result in crash
7. Using the Difference option, click the Difference Button
8. *This is where the crash occurs*

### Actual results

The application crashes and closes

### Expected results

The mesh boolean to occur without crashing the application

### Project file & Debug log uploads

[RepRack.zip](https://github.com/user-attachments/files/19663559/RepRack.zip)
[log.zip](https://github.com/user-attachments/files/19663560/log.zip)

### Checklist of files to include

- [x] Log file
- [x] Project file

Haidiye00, 2025-04-09:
Okay, we will conduct an analysis~

Haidiye00, 2025-04-10:
[image]
Hello, our internal version has been fixed and will be included in the next version~

seanbo, 2025-04-10:
Fantastic. Thank you.

closed 2025-04-10 by seanbo

FabianNiesen, 2026-04-16:
I have a similar issue, when will the new version released or is there a beta? @Haidiye00