bambulab/BambuStudio #6477: Filament keeps changing back to PLA

nagyfrantisek, 2025-04-09:
### Bambu Studio Version

2.0.1.50

### Where is the application from?

Bambu Lab Official website

### OS version

macOS 15.4

### Printer

BambuLab P1S

### How to reproduce

When synchronizing the filament settings in Bambu Studio, the software sets the filament to material, which is currently loaded in the printer. Now I have PETG there. However, if I change the layer height, the filament setting reverts back to Generic PLA.

**Steps to Reproduce:**

1. Open Bambu Studio.
2. Synchronize the filament settings with the printer.
3. Observe that the filament is set to PETG.
4. Change the layer height in the settings.
5. Notice that the filament setting changes back to Generic PLA.

https://github.com/user-attachments/assets/ac9edfe0-6292-4808-8fa1-0a4b5e777950

### Actual results

The filament setting changes back to Generic PLA.

### Expected results

The filament setting should remain as PETG after changing the layer height.

### Project file & Debug log uploads

-

### Checklist of files to include

- [ ] Log file
- [ ] Project file

BoBoYu-1, 2025-04-10:
Thank you for your feedback. Could you please provide the log file so that we can solve the problem?

nagyfrantisek, 2025-04-10:
> Thank you for your feedback. Could you please provide the log file so that we can solve the problem?

Log file attached to bug report

nagyfrantisek, 2025-04-20:
The bug still occurs with version 2.0.2.57