bambulab/BambuStudio #6478: Incorrect Temperature Settings for PETG in Bambu Studio

nagyfrantisek, 2025-04-09:
### Bambu Studio Version

2.0.1.50

### Where is the application from?

Bambu Lab Official website

### OS version

macOS 15.4

### Printer

BambuLab P1S

### How to reproduce

After setting the material to PETG in Bambu Studio and slicing the model, the print job is sent to the printer with the correct PETG material setting. However, when the printer starts, it uses a nozzle temperature of 220°C and a bed temperature of 55°C, which are the default settings for generic PLA instead of the appropriate settings for PETG.

**Steps to Reproduce:**

- Open Bambu Studio.
- Set the material to PETG.
- Slice the model.
- Send the print job to the printer, which is also set to use PETG.
- Start the print job.
- Observe the nozzle and bed temperatures during printing.

### Actual results

The printer uses PLA settings (220°C for the nozzle and 55°C for the bed) instead of the correct PETG settings.

### Expected results

The printer should use the correct temperature settings for PETG which is 250°C for the nozzle and 70°C for the bed.

### Project file & Debug log uploads

-

### Checklist of files to include

- [ ] Log file
- [ ] Project file

Bestivus, 2025-04-09:
Seeing the same, version 2.0.1.50.  Bambu PETG HF white, printing from A1.

Went to start a print last night and noticed 50% through that bed adhesion was awful - edge curling, shifting layers, and stringing.  Thought the bed needed to be cleaned, so washed with Dawn dish soap, dried, and rubbed down with IPA.  

Started the job again and noticed at around layer 7 that the bed temperature was 55°C and nozzle temp was 220°C.  Manually increased both to what they should have been, and the project completed, but it still had visible signs of the initial lower temps.

GuyH77, 2025-04-10:
Definitely a bug. I'm also seeing extremely slow print times compared to Bambu PETG-HF. I'm not sure why though. I need to investigate more. Walls printing at 52mm/s

nagyfrantisek, 2025-04-20:
The bug still occurs with version 2.0.2.57

JohnSmithCYM, 2025-04-22:
@nagyfrantisek @Bestivus @GuyH77 @jwise I have followed the the steps @nagyfrantisek and it can't reproduce this question.
[image]
I am trying to figure out why this happend and can anyone upload something here or if send e-mail to malkey.chen@bambulab.com(we promise this will be no data leak and only use for this is to find out the problem):
- Opeartions in video cut. 
- Wrong gcode.3mf/gcode and note the original profile you use.
- The log file package.
Thanks for you all again, and we wish your feedback.

nagyfrantisek, 2025-04-24:
@JohnSmithCYM  I can't always reproduce the bug. I made two videos about the problem. In the [first video](https://www.youtube.com/watch?v=oUg5uQGhEr4), you can see that PETG is set everywhere, but when the printing starts, the temperature shows that it is using PLA temperatures for some reason. It seems like the slicer is accidentally used the PLA profile for slicing.
The [second video](https://www.youtube.com/watch?v=28NH1wrNqqw) was taken 3 minutes later, and I didn't change anything in the settings; everything remained set to PETG. Interestingly, when the printing starts, it prints perfectly at the PETG profile temperature.

[debug_Thu_Apr_24_19_05_35_45935.log.0.zip](https://github.com/user-attachments/files/19896413/debug_Thu_Apr_24_19_05_35_45935.log.0.zip)

Haidiye00, 2025-04-25:
Hello, please use the latest version
https://github.com/bambulab/BambuStudio/releases/tag/V02.00.03.54

JohnSmithCYM, 2025-04-27:
@nagyfrantisek @Bestivus @GuyH77 Is thers any incorrect temperature settings failure after updated to V2.0.3.54?