bambulab/BambuStudio #6479: Studio produces warning when using Support for PLA/PETG and PETG filliment

Raelx, 2025-04-09:
### Bambu Studio Version

2.0.1.50

### Where is the application from?

Bambu Lab Official website

### OS version

windows 11 24H2

### Printer

X1C + AMS

### How to reproduce

Enable supports and select PETG HF as the main filament and "Support for PLA/PETG" as the support interface material. All the filaments I have are official Bambu filament. Slice the plate.

### Actual results

Studio will produce a warning about mixing PLA and PETG.

### Expected results

Given that the support material is specifically sold for use with PETG this error warning seems incorrect. I do not believe there was a warning in the same situation before BBS version 2

### Project file & Debug log uploads

[image]

### Checklist of files to include

- [ ] Log file
- [ ] Project file

BoBoYu-1, 2025-04-10:
Thank you for your feedback. This warning does not affect printing. However, if PLA consumables and PETG consumables are used for printing together and the parameters are not adjusted properly, it will affect the printing quality. Therefore, this warning is given to facilitate parameter adjustment and reduce printing quality issues.

Raelx, 2025-04-10:
I understand the warning but the reason I'm buying RFID Bambu filament is so I don't have to adjust any parameters and I count on BBS to do this automatically, which I believe it's currently doing for this support material with PETG. This is why I believe this warning is in error.
Should this warning be show if someone is using PLA as support interface, yes. But when using BBL filaments designed and sold to be used together BBS should be managing all the parameters and not producing warning.

JohnSmithCYM, 2025-04-16:
@Raelx Thanks for your feedback, and we discuss two conditions that this warning could be optimized:
- When PETG and PLA exist on plate both as single object, the warning should show off.
- When PETG was used as support material for PLA, or reverse, we can use the best parameter(like what Spirial Vase or SupPLA does) for user.

bobotu, 2025-04-23:
But in this case, I'm going to use **Support for PLA/PETG** to support **PETG** object. Isn't the support filament designed for **PETG**? I assume this support filament is **compatible** with PETG.

WilliamBurkert, 2025-05-26:
[image]

longrackslabs, 2025-06-29:
> Thank you for your feedback. This warning does not affect printing. However, if PLA consumables and PETG consumables are used for printing together and the parameters are not adjusted properly, it [...]

This is fine, but once I go the the wiki and get things setup, then I don't need Studio to keep reminding me.

longrackslabs, 2025-06-29:
> I understand the warning but the reason I'm buying RFID Bambu filament is so I don't have to adjust any parameters and I count on BBS to do this automatically, which I believe it's currently doing [...]

It should show the warning once, and there should be way to suppress this warning.