bambulab/BambuStudio #6480: Progress bar is missing

Astroskg, 2025-04-10:
### Bambu Studio Version

2.0.1.50

### Where is the application from?

Bambu Lab Official website

### OS version

Mac OS 15.3.1 

### Additional system information

16 GB
M-1 core 
2020 Mac mini

### Printer

Bambu labs X1-Carbon, Bambu labs P1S 

### How to reproduce

go to multi device 
click on tc
machine lists your printers 
in Device status window it used to show the print job progress 
now it only says Printing as if i didn't already know that 

### Actual results

Device status only shows "Printing" not the progress of the job like it used to 

### Expected results

i'd like to get the progress bar back , it really helps to be able to go to  single screen nd see your printers and know where they are at in the print process without having to open Bambu studio's or bambu handy to see it still has XXXX time left 

### Project file & Debug log uploads

i do not have a file or screen pic, this problem is easily duplicated bu just clicking the tab

### Checklist of files to include

- [ ] Log file
- [ ] Project file

AmDDJunkie, 2025-04-10:
Also getting this result on Windows 11.

StineDEk, 2025-04-15:
I would like to have the progress bar back as well, please!

walterwongbbl, 2025-04-21:
Hi. @Astroskg @AmDDJunkie 
This should be a functionality regression, and we have added it to the fix list. Thank you very much.