bambulab/BambuStudio #6481: multicolor

bbrady1221, 2025-04-10:
### Bambu Studio Version

ffrf

### Where is the application from?

Bambu Lab Official website

### OS version

sdfgerdgre

### Printer

bambu a1

### How to reproduce

multicolor does not work fix it!!!!

### Actual results

no go f yopur self

### Expected results

fix multicolor

### Project file & Debug log uploads

awergasredg

### Checklist of files to include

- [ ] Log file
- [ ] Project file

BoBoYu-1, 2025-04-10:
Sorry, could you please provide a detailed description of the screen recording to explain the problem you encountered?

BoBoYu-BamBu, 2025-04-28:
@bbrady1221 Hello, have your questions been resolved?