bambulab/BambuStudio #6484: Studio showing "Online Models" in home screen, even when option to show online models is disabled in preferences

rafabe77, 2025-04-10:
### Bambu Studio Version

2.0.1.50

### Where is the application from?

Bambu Lab Official website

### OS version

Windows 10 22H2

### Additional system information

There is an option in Bambu Studio preferences that says:
Online Models
- Show online staff-picked models on the home page
- show history on the home page

Both options are disabled, however, the home page still shows the online models.

Reason for this report:
I'm trying to convince my company to buy an H2D for prototyping, but doesn't look professional when I open the software and the first thing it shows is a 3D printed Pikachu or a fidget.

### Printer

H2D

### How to reproduce

1. Open Bambu Studio, the home page shows the online models
2. Ctrl+P for Preferences, Scroll down until "Online Models"
3. Remove the tick mark on "Show online staff-picked models on the home page"
4. Close the preferences. The online models are still there.
5. Close Bambu Studio and open it again. The online models are still there.

[image]

[image]

### Actual results

The online models are still showing on the home page

### Expected results

The home page NOT showing online models

### Project file & Debug log uploads

[log.zip](https://github.com/user-attachments/files/19693900/log.zip)

### Checklist of files to include

- [x] Log file
- [ ] Project file

pgifford, 2025-07-17:
@MackBambu You picked up #6376, this is the same. I was disappointed to see this wasn't addressed in the beta that just came out.

MackBambu, 2025-09-05:
fixed:
https://github.com/bambulab/BambuStudio/commit/e18b5384b2b156f444a78f8fde0c02782a5a40ec

closed 2025-09-05 by MackBambu