bambulab/BambuStudio #6486: Problem with the calibration of the flow dynamic

McUser95, 2025-04-11:
### Bambu Studio Version

2.0.2.50

### Where is the application from?

Bambu Lab Official website

### OS version

macOS Sequoia 15.3.2

### Printer

Bambu Lab X1C

### How to reproduce

1. Go to calibration of the flow dynamic

### Actual results

When I want to calibrate several filaments, “Incompatible” appears under the AMS for the inserted filaments

### Expected results

The inserted filament defined in the X1c should be visible there.

### Project file & Debug log uploads

[image]

### Checklist of files to include

- [ ] Log file
- [ ] Project file

ProTech2k5, 2025-04-15:
I had this issue as well on my A1, ended up installing a previous ver. of bambu studio and downgrading the printer/ams firmware down 2 versions

zhimin-zeng-bambulab, 2025-04-21:
@McUser95  Could you show me your filament settings on the device page?