bambulab/BambuStudio #6488: MQTT Error

AFKreativ, 2025-04-11:
### Bambu Studio Version

2.01.50

### Where is the application from?

Bambu Lab Official website

### OS version

Windows 10

### Printer

Bambulab X1 Carbon 01.08.05.00

### How to reproduce

HMS Code HMS_0500-0500-0001-0007
The error occurs since the last update of studio

### Actual results

Control of the printer from the studio is not possible

### Expected results

According to Bambuwiki, the error should not occur with the versions of Studio and Printer

### Project file & Debug log uploads

[log.zip](https://github.com/user-attachments/files/19706151/log.zip)

### Checklist of files to include

- [x] Log file
- [ ] Project file

BoBoYu-BamBu, 2025-04-15:
Thank you for your feedback. After our investigation, it turns out that not only is MQTT connection failed on your end, but other network services have also failed. Could you try another network connection?

nleborgne, 2025-09-09:
I am encountering the same issue