bambulab/BambuStudio #6494: Projekteinstellungen werden nicht übernommen

alfameanchen, 2025-04-11:
### Bambu Studio Version

2.0.1.5

### Where is the application from?

Bambu Lab Official website

### OS version

Mac10.5

### Printer

A1 und A1 mini

### How to reproduce

Profileinstellungen meiner einzelnen Projekte werden nicht übernommen. Beim Öffnen von 2 Projekten nach einander wird immer das vorherige Profil übernommen. Alle Einstellungen wie Stützstruktur, A1 oder A1Mini Auswahl, Filamentprofil und Stärke werden vom vorherigen automatisch  übernommen, obwohl ich das nicht ausgewählt habe oder bestätige. Das ist extrem frustrierend alle Einstellungen neu durchzuführen. 

### Actual results

Profileinstellungen meiner einzelnen Projekte werden nicht übernommen. Beim Öffnen von 2 Projekten nach einander wird immer das vorherige Profil übernommen. Alle Einstellungen wie Stützstruktur, A1 oder A1Mini Auswahl, Filamentprofil und Stärke werden vom vorherigen automatisch  übernommen, obwohl ich das nicht ausgewählt habe oder bestätige. Das ist extrem frustrierend alle Einstellungen neu durchzuführen. 

### Expected results

Projekteinstellungen vom jeweiligen Projekt. 

### Project file & Debug log uploads

Profileinstellungen meiner einzelnen Projekte werden nicht übernommen. Beim Öffnen von 2 Projekten nach einander wird immer das vorherige Profil übernommen. Alle Einstellungen wie Stützstruktur, A1 oder A1Mini Auswahl, Filamentprofil und Stärke werden vom vorherigen automatisch  übernommen, obwohl ich das nicht ausgewählt habe oder bestätige. Das ist extrem frustrierend alle Einstellungen neu durchzuführen. 

### Checklist of files to include

- [ ] Log file
- [ ] Project file

alfameanchen, 2025-04-11:
ich will unbedingt downgraden auf die alte Bambu Studio Slicer Version davor. Ich brauche diese Datei bitte zum download. Die aktuelle funktioniert nicht.

BoBoYu-BamBu, 2025-04-15:
Hier ist die Übersetzung ins Deutsche:
"Vielen Dank für Ihr Feedback. Könnten Sie bitte einen Bildschirm aufnehmen, um Ihr Problem zu beschreiben, damit wir es nachvollziehen und lösen können? Bitte stellen Sie uns zusätzlich die Logdateien zur Verfügung."