bambulab/BambuStudio #6495: H2D - Support for different nozzle width

imaginedvl, 2025-04-11:
**Is your feature request related to a problem? Please describe.**
Right now a lot of us are using the 0.2 to print very detailed text on top (or the bottom) of the part.
And this is usually useful for only few layers (maybe up to 1 mm with several layer) but then the body of the part can be printed in 0.4 or even higher.

**Describe the solution you'd like**
 If the slicer would consider the different nozzle type, we could print the detailed part using the 0.2 nozzle and the rest in 0.4/0.6/0.8 making the print time considerably smaller.

**Describe alternatives you've considered**
Alternative is to print the whole part in 0.2.

**Additional context**
NA

abudden, 2025-06-11:
+1 for this.  Even if it were restricted to only work layer-by-layer (i.e. either the first or last couple of layers would be with the smaller nozzle) then it would work for most use cases.  I can imagine that making it work with swapping nozzle size at any point in the print could be a lot more complicated.

gshort, 2025-06-30:
Another +1, this would fit a current use case exactly, and would make the dual nozzles of the H2D significantly more useful. Working on a project right now where printing the lower structural portion with a 0.8 nozzle and then switching to a fine 0.2 nozzle for the upper cosmetic portion would be an absolute godsend and would save countless hours of print time with no loss of quality.

B-S-B, 2026-10-02:
Another +1
This i will make the H2D a dual nozzel printer in true sense. This has been a long pain point for adding text on top/bottom of any print. Fine text usually come out bad.

heinemil, 2026-10-30:
+1. Dual nozzles should offer more than faster colour changes and reduced purging. Printing the main body with a 0.6 mm nozzle, then switching to 0.2 mm for detailed lettering, would make the H2D substantially more useful—even without mixing nozzle sizes within the same layer.

Customers should not have to print an entire large part through a 0.2 mm nozzle just because a few layers need fine details. This is exactly the kind of capability that helps justify choosing a premium dual-nozzle printer over a cheaper single-nozzle model.

OrcaSlicer 2.5.0 alpha already supports mixed nozzle sizes on the H2D/X2D. Why should customers need third-party software to access this capability on Bambu’s own hardware?

Please provide a concrete development update and target release window. Layer-by-layer support would be a valuable first step.