bambulab/BambuStudio #6496: Excessive Prime Tower and Flushing Amounts

seanbo, 2025-04-11:
### Bambu Studio Version

2.0.1.50

### Where is the application from?

Bambu Lab github releases

### OS version

Windows 11

### Additional system information

CPU: Ryzen 5
RAM: 64GB DDR5
GPU: Nvidia 4090ti

### Printer

Bambu Lab A1 & A1 Mini

### How to reproduce

1. Selects Plate 3
2. Click Slice
3. Select Filament in drop down

### Actual results

There is an excessively large prime tower generated.  The model itself uses less than 4g of filament, but the prime tower and flushing is several hundred.  I made a copy of the .3mf file, repainted the model with the same colors, re-sliced it and got something more inline with expectations.

Files Included:
Multi Color Stitch-BambuStudio.3mf - This is the file with the "issue"
Multi Color Stitch-BambuStudio-second_try.3mf - This is the file that seems normal

[image]

### Expected results

the same volumes as found in "Multi Color Stitch-BambuStudio-second_try.3mf"

### Project file & Debug log uploads

[log.zip](https://github.com/user-attachments/files/19713758/log.zip)

[Multi Color Stitch-BambuStudio.zip](https://github.com/user-attachments/files/19713768/Multi.Color.Stitch-BambuStudio.zip)

[Multi Color Stitch-BambuStudio-second_try.zip](https://github.com/user-attachments/files/19713769/Multi.Color.Stitch-BambuStudio-second_try.zip)

### Checklist of files to include

- [x] Log file
- [x] Project file

Guenni1975, 2025-04-28:
Unfortunately, I can still confirm the problem in version 2.0.3.54.
The Prime Tower is huge and unfortunately cannot be customised.

[image]

Guenni1975, 2025-05-05:
I found out that the problem with the size of the Prime Tower only occurs when I print several multi-coloured objects on one plate. The Prime Tower increases in size with each object!

zhimin-zeng-bambulab, 2025-05-06:
1. You can turn off the rib wall and adjust the infill gap to 100%, which will make the tower more compact.
[image]

2. The original prime volume has been changed to a filament parameter. You can adjust the size of the prime tower by adjusting this parameter.
[image]

seanbo, 2025-05-07:
> 1. You can turn off the rib wall and adjust the infill gap to 100%, which will make the tower more compact.
>    [image]
> 2. The original prime volume has been changed to a filament parameter. You can adjust the size of the prime tower by adjusting this parameter.
>    [image]

Did you see that I resliced the same file after repainting (same filament and colors, just painted it again in the same places) and the prime tower changed?