bambulab/BambuStudio #6498: Connecton become fail to Printer X1 Carbon

inforlab, 2025-04-11:
### Bambu Studio Version

2.0.1.50

### Where is the application from?

Bambu Lab Official website

### OS version

MacOsX Ventura 13.7.5

### Additional system information

CPU : Intel 3,4 GHz Intel Core i5 quatre cœurs
RAM : 24Go 2400 MHz DDR4
GPU : Radeon Pro 570 4 Go

### Printer

BambuLab X1 Carbon AMS

### How to reproduce

Start Printer
Start Bambu Studio 
The printer appears
After sending to printer , a message appears on the left upper corner "Fail to connect printer" with a red cross

[image]

### Actual results

No results

### Expected results

Nothing

### Project file & Debug log uploads

??

### Checklist of files to include

- [ ] Log file
- [ ] Project file

JonyRalf, 2025-04-14:
I fully support this bug for A1. 

For Apple Mac (m1), Windows, phone

[image]
[image]
[image]
[image]

BoBoYu-BamBu, 2025-04-15:
Thank you for your feedback. After our investigation, it turns out that we are unable to connect to our MQTT server. Are you using a VPN or a proxy? If so, please turn it off and try again.

BoBoYu-BamBu, 2025-04-15:
Could you check on Wiki to see if there are any ports blocked by the firewall? https://wiki.bambulab.com/en/general/printer-network-ports

inforlab, 2025-04-15:
> JonyRalf
>  left a comment
> (bambulab/BambuStudio#6498)
> There are no problems with the firewall. The most interesting thing is that the connection works via LAN for A1 - it just doesn't want to connect via Wi-Fi.
>
> VPN is also disabled

All issues appeared from the latest version of Bambustudio (2.0.1.50)

walterwongbbl, 2025-04-16:
Hi @inforlab 
Can you provide us with your studio logs? We would like to analyze the reasons.

dyp, 2025-04-19:
[debug_Sat_Apr_19_09_31_49_17856.log.zip](https://github.com/user-attachments/files/19818581/debug_Sat_Apr_19_09_31_49_17856.log.zip)

I have the same problem with X1C. Logs are attached.

eyast, 2025-04-21:
Same is happening for me - I can't print. I run the latest version of bambu studio 2.0.2.57 and X1c used to work fine (01.08.05.00 with AMS 00.0006.44). For the last couple of days the printer has been showing "MQTT verfiication failed". connected to WiFi (not LAN, not developer mode).
log files of Bambu studio:

[debug_network_Mon_Apr_21_10_01_30.log.zip](https://github.com/user-attachments/files/19827725/debug_network_Mon_Apr_21_10_01_30.log.zip)

eyast, 2025-04-21:
After restarting the app, the printer, the router, something magically now works. it prints. I suspect the problem must be local to my end, but I can't pinpoint what happens, or why it suddenly works. I think my network must be at fault someway, somehow.