bambulab/BambuStudio #6502: Changing layer filament sequence almost doubles print time

Festivejelly, 2025-04-11:
### Bambu Studio Version

2.0.1.50

### Where is the application from?

Bambu Lab github releases

### OS version

Windows 10

### Printer

H2D

### How to reproduce

Import an model made of 2 objects
Set 1 objects as filament 1
Set 2nd object as filament 2

Click slice. Print time looks normal.

If you then click the plate settings and re order the filaments:

[image]

The print time drastically goes up, I cant see a reason as to why? It also seems to print a much bigger purge tower.

### Actual results

[image]

[image]

### Expected results

[image]

[image]

### Project file & Debug log uploads

Proprietary file

### Checklist of files to include

- [ ] Log file
- [ ] Project file

BoBoYu-BamBu, 2025-04-15:
Thank you for your feedback. This printing time is normal. The customized grouping you made will result in many more head-changing times. If you want to save the printing time, just use the default automatic one directly.

BoBoYu-BamBu, 2025-04-28:
@Festivejelly  Hello, do you have any further questions regarding this issue? If not, can we close this issue now?

closed 2025-04-29 by BoBoYu-BamBu