bambulab/BambuStudio #6504: Changing Font without Unselecting Causes Font to Move Locations

chriskewl, 2025-04-11:
### Bambu Studio Version

Latest

### Where is the application from?

Bambu Lab Official website

### OS version

Windows 10

### Printer

X1C

### How to reproduce

- Click on Top Tag text
- Change letters
- Click on Bottom Tag text
- Change letters, this will cause the text to move or error out.

### Actual results

This will cause the text to move or error out IF you don't unselect the assembly THEN click on Bottom Tag and modify it.

### Expected results

It should just allow the text to be edited freely.

### Project file & Debug log uploads

[31.5mm+Round-Back-Writing.zip](https://github.com/user-attachments/files/19714969/31.5mm%2BRound-Back-Writing.zip)

### Checklist of files to include

- [ ] Log file
- [x] Project file

Haidiye00, 2025-04-13:
Hello, can you download ScreenToGif software and record a GIF of the operation video and upload it? Thank you
[Release ScreenToGif 2.41.1 · NickeManarin/ScreenToGif](https://github.com/NickeManarin/ScreenToGif/releases/tag/2.41.1)

nefas83, 2025-05-07:
I have the same bug.
Here is the gif you requested.

[image]

closed 2025-07-22 by Haidiye00