bambulab/BambuStudio #6506: I am insanely upset.  All of my Project files have been deleted from my computer ( MacBook Pro ).  They are the only files missing,  This can't be blamed on another program.

kcclark, 2025-04-12:
### Bambu Studio Version

2.0.1.50

### Where is the application from?

Bambu Lab Official website

### OS version

15.3.2 (24D81)

### Additional system information

MacBook M3 Pro 18GB 

### Printer

Bambu Labs P1S 

### How to reproduce

I have no idea.

### Actual results

The is criminal malice or negligence.

### Expected results

The is criminal malice or negligence.

### Project file & Debug log uploads

The is criminal malice or negligence.

### Checklist of files to include

- [ ] Log file
- [ ] Project file

kcclark, 2025-04-12:
The is criminal malice or negligence.

kcclark, 2025-04-12:
[image]

Haidiye00, 2025-04-13:
Hello, can you download QuickRecorder software and record a GIF of the operation video and upload it? Thank you                         [lihaoyun6/QuickRecorder: A lightweight screen recorder based on ScreenCapture Kit for macOS / 基于 ScreenCapture Kit 的轻量化多功能 macOS 录屏工具](https://github.com/lihaoyun6/QuickRecorder/releases/tag/1.6.5)

davidjdixon, 2025-04-13:
I'm not sure how you're setup, but /private/var/folders (or the /var/folders symlink) is not a location you should be saving your documents to, and definitely is not a default location of Bambu Studio to save documents. Most likely, it's inheriting the last save location from another program on your Mac since it utilises the macOS save dialog.

closed 2025-07-17 by Haidiye00