bambulab/BambuStudio #6507: Duplicating STL files in 3mf project

J-Corrall, 2025-04-12:
### Bambu Studio Version

2.0.1.50

### Where is the application from?

Bambu Lab Official website

### OS version

Windows 11

### Additional system information

CPU: AMD Ryzen 9 3900X 12-Core Processor (24 CPUs), ~4.1GHz
RAM 32GB 3600 MT/s
Graphics: NVIDIA GeForce RTX 3070 16GB

### Printer

Bambu 

### How to reproduce

- Create a 3MF file
- Add multiple .stl files to multiple build plates
- Save file 
- Close Bambu Studio
- Open Bambu Studio
- Attempt to slice files on one of the build plates 
- Error shows that files are overlapping, STL files are duplicated, two files (same file) appearing superimposed one on top of the other
- If duplicate files are deleted, the project is saved, Bambu Studio is closed and reopened, the duplicate files reappear

### Actual results

Files appear to duplicate as described above

### Expected results

Files should not duplicate after saving. 

### Project file & Debug log uploads

[CubePro Store Cubes.zip](https://github.com/user-attachments/files/19719221/CubePro.Store.Cubes.zip)

### Checklist of files to include

- [ ] Log file
- [x] Project file

Haidiye00, 2025-04-13:
[image]
Hello, after importing the file, I noticed that several STL files may overlap. You can manually adjust their spatial positions or use the automatic disk placement feature~
But I didn't reproduce the repetition issue you mentioned.
can you download ScreenToGif software and record a GIF of the operation video and upload it? Thank you
[Release ScreenToGif 2.41.1 · NickeManarin/ScreenToGif](https://github.com/NickeManarin/ScreenToGif/releases/tag/2.41.1)

qbbosfab, 2025-04-13:
I am having the same issue....I have not quite narrowed down how to reproduce it but it is happening quite often since updating to 2.0.1.50

deletedalien, 2025-05-19:
Hi there i just recentl;y noticed this started to hapen as well...

Haidiye00, 2025-06-13:
https://github.com/bambulab/BambuStudio/releases/tag/v02.01.00.59
Hello, please upgrade to the latest version and see if there are still any issues. Thank you
And can you download ScreenToGif software and record a GIF of the operation video and upload it? Thank you
[Release ScreenToGif 2.41.1 · NickeManarin/ScreenToGif](https://github.com/NickeManarin/ScreenToGif/releases/tag/2.41.1)

Haidiye00, 2025-08-01:
https://github.com/bambulab/BambuStudio/releases
Please use the latest version, thank you
If there are any further questions, we will continue to discuss

closed 2025-08-01 by Haidiye00