bambulab/BambuStudio #6509: printing speed on overhangs

3D-Michael, 2025-04-13:
### Bambu Studio Version

2.0.1.50

### Where is the application from?

Bambu Lab github releases

### OS version

Ubuntu 24.04

### Additional system information

CPU: N100
RAM: 16GB
Intel-Graphic

### Printer

Bambu X1C

### How to reproduce

I have a part with slight overhangs and disabled the "Slow Down on Overhangs" feature. Even so, the printer still significantly reduces its speed.

### Actual results

The printer significantly reduces its speed.

### Expected results

The speed should not be reduced if I disable the "Slow down on overhangs" function.

### Project file & Debug log uploads

[Dose-H66.zip](https://github.com/user-attachments/files/19723285/Dose-H66.zip)

### Checklist of files to include

- [ ] Log file
- [ ] Project file

davidjdixon, 2025-04-13:
I can also confirm this occurs and have seen the effect using the H2D on vase mode prints with significant perimeter overhangs.

QingZhangBambu, 2025-04-21:
these parts are overhang wall，which detected by  "Detect overhang walls" function,  not affacted by "Slow Down on Overhangs" feature

[image]

 
[image]