bambulab/BambuStudio #6511: Remembering last used filament instead of filament profile selected and silced

btalbot85, 2025-04-13:
### Bambu Studio Version

2.0.1.50

### Where is the application from?

Bambu Lab Official website

### OS version

Windows 11

### Printer

X1C With AMS

### How to reproduce

Print in selected filament.
Switch filament and print another.

### Actual results

When printing from one filament to another the last used filament profile continues to upload even when the new filament is clearly selected.
For example (And this caused an hour of disassembly removing the clog/blockage it caused) I printed a model in PA6, then selected PETG, resliced and sent to the printer. I didnt notice but it was printing the PA6 temps, confirmed my PETG profile wasnt changed or anything. The high temps caused a horrible tangle in the extruder taking forever to take it all apart and fix.
Once fixed was able to reproduce with multiple filament profiles. Only way to work around is to close and reopen bambu studio. Problem just started when I updated to this version.

### Expected results

Filament profile should switch correctly and proper temps used.

### Project file & Debug log uploads

None

### Checklist of files to include

- [ ] Log file
- [ ] Project file

Haidiye00, 2025-04-16:
Hello, please use the latest version, thank you.
https://github.com/bambulab/BambuStudio/releases/tag/v02.00.02.57

JohnSmithCYM, 2025-04-23:
Can you offer your 'wrong'(we want to see what happened in the specific gcode) gcode.3mf or gocde to us. Thanks for your feedback again.

JohnSmithCYM, 2025-05-15:
Have you encountered this problem after you update to V2.0.3.54(https://github.com/bambulab/BambuStudio/releases/tag/V02.00.03.54)?We have fix a similar problem in this version.