bambulab/BambuStudio #6512: Problem with gap fill on top surface layers

condes777, 2025-04-13:
Hi, I’m a regular user of Bambu Lab printers and I really appreciate all the work your team is doing to improve the printing experience. I wanted to share something I’ve noticed that’s affecting the final quality of some prints—specifically, the way gap fill works on top surfaces.

Near the end of a print, when the printer starts filling tiny gaps on the top layers, the nozzle passes over an already finished surface. Because that gap fill is so thin, the hot nozzle ends up slightly deforming the top layer. The result doesn’t look clean—it loses that crisp, smooth finish we usually aim for.

I’ve tried ironing as an alternative, but I’m not a fan of how it changes the texture. It feels too artificial, and I personally prefer a more natural surface finish without that “ironed” look.

That’s why I think it would be super helpful to have an option to disable gap fill on top layers. It would give users more control over the final appearance, especially for aesthetic or detailed prints.

Thanks for taking the time to read this, and again, great job with everything so far. Looking forward to future updates!

[image]

[image]
[image]
[image]
[image]
[image]

condes777, 2025-04-13:
After sharing the images, it becomes clear how the issue is concentrated in specific areas—like the bottom-left corner of one of the printed parts. In that spot, the nozzle attempted to perform gap fill, but the result was visible surface deformation. That tiny action ended up damaging the print, creating a melted or warped appearance.

This kind of fill offers no meaningful functional or visual benefit. In fact, the final result without the fill is much cleaner and more natural. We're not talking about large gaps or structural weaknesses—these are minimal areas that, once touched by the hot nozzle, end up compromising the quality of the print.

By reviewing the slicer in Bambu Studio, it’s easy to see that the deformed zones line up precisely with the gap fill toolpaths. That pattern led me to dig deeper, but unfortunately I couldn’t find any option to disable it. I also couldn’t find any clear guidance in the official documentation or within the Spanish-speaking community.

That’s why I decided to submit this report. I'm not interested in using ironing as a workaround, since I don’t like the resulting texture. What I really need is an option to disable top layer gap fill, without applying broader surface effects that change the print’s overall look.

I’m part of a Spanish-speaking community, and I know others have noticed this too—even if few have taken the step to report it. I hope this information is useful and can help improve the printing experience even more.

condes777, 2025-04-13:
I’ve also tried using the developer mode tools, but I couldn’t find any setting that allows changing this behavior. There’s no available option to disable gap fill on the top surface layers. I’ve gone through previous posts from Bambu Lab and explored advanced settings, but there’s no reference or control for this specific issue.

Even Bambu Studio itself indicates that disabling the “fill gaps” option only affects internal layers, not top or bottom surfaces. So, from what I understand, there’s currently no way for users to manage this setting directly.

That’s why I suggest making this feature configurable—so we can manually enable or disable it depending on the finish we want to achieve. Having that level of control would help improve surface quality without relying on solutions like ironing, which don’t always provide the aesthetic we’re aiming for.

frasier2k, 2025-04-17:
It's already opend here.
https://github.com/bambulab/BambuStudio/issues/4090

condes777, 2025-04-22:
> It's already opend here. [#4090](https://github.com/bambulab/BambuStudio/issues/4090)

Yes, that's correct, but the bambulab development team hasn't provided any solutions. It seems the Orca Slicer team isn't wasting time and making changes faster.