bambulab/BambuStudio #6514: Something is wrong with PET-CF profile?

pompushko, 2025-04-13:
### Bambu Studio Version

2.0.1.50

### Where is the application from?

Bambu Lab github releases

### OS version

Windows 10

### Printer

X1C

### How to reproduce

Im using Fiberon PET-CF integrated profile. But it prints with 220c temp...

What Im doing wrong?

### Actual results

[image]

[image]

### Expected results

Print 300c as was before.

### Project file & Debug log uploads

[Body20.zip](https://github.com/user-attachments/files/19726309/Body20.zip)

### Checklist of files to include

- [ ] Log file
- [ ] Project file

JohnSmithCYM, 2025-04-22:
@pompushko We have tested your 3mf and we can't reproduce your problem. Can you offer your 'wrong'(we want to see what happened in the specific gcode) gcode.3mf or gocde to us. Thanks for your feedback again.

JohnSmithCYM, 2025-04-22:
may be the same question with #6197

XunZhangBambu, 2025-05-13:
@pompushko  Out latest version have fixed the problem. You can download and have a try

closed 2025-05-13 by XunZhangBambu