bambulab/BambuStudio #6516: https://midi-to-gcode.bambulab.com/edit can not hear any sond

thelastidea, 2025-04-14:
### Bambu Studio Version

1.6

### Where is the application from?

Bambu Lab Official website

### OS version

win10

### Printer

h2d

### How to reproduce

use edge open https://midi-to-gcode.bambulab.com/edit.

### Actual results

i can not hear any sond,only this website has this problem,i use another computer but it is still.

[image]

### Expected results

i think i should hear something

### Project file & Debug log uploads

.......

### Checklist of files to include

- [ ] Log file
- [ ] Project file

BoBoYu-BamBu, 2025-04-15:
Could you elaborate on the problem you have encountered?

BoBoYu-BamBu, 2025-04-15:
Could you elaborate on the problem you have encountered? I noticed that your version is a bit outdated. Please update to the latest version V2.0.1.50 on our official website. The download link is：https://bambulab.cn/zh-cn/download/studio

BoBoYu-BamBu, 2025-04-28:
@thelastidea  Hello, have your questions been resolved?