bambulab/BambuStudio #6521: cant log in

habloola, 2025-04-14:
### Bambu Studio Version

2.00.01.50

### Where is the application from?

Bambu Lab Official website

### OS version

windows 10.0.26100

### Additional system information

[image]

### Printer

bambulab a1

### How to reproduce

go to the app
press login

### Actual results

blank page opens and i cant log in

### Expected results

i need to log in

### Project file & Debug log uploads

?

### Checklist of files to include

- [ ] Log file
- [ ] Project file

BoBoYu-BamBu, 2025-04-15:
Thank you for your feedback. Could you try again with a different network?

habloola, 2025-04-15:
Hi
I saw on github that when i choose language in settings this is solved and
so it did

דניאל טביב :)

BoBoYu-BamBu, 2025-04-15:
Hello，OK，Is it okay to close this issue report now that your problem has been solved?

realzorro, 2025-04-21:
We have identified this issue, please stay tuned for future releases of new versions.
At present, you can configure the language in the settings interface to try and solve the problem.