bambulab/BambuStudio #6522: Cannot login - App crash

polowiper, 2025-04-14:
### Bambu Studio Version

1.10.2.76

### Where is the application from?

Built from source by yourself or others

### OS version

Nixos 25.05

### Printer

Bambulab A1

### How to reproduce

1.Open Bambustudio
2.Try to log in using any of the method
3.Login with other accounts fails and login with email and password makes the app crashes

### Actual results

Login with other accounts fails (goes to this [link)](https://bambulab.com/en-us/sign-in/studio-callback?result=fail&error=get_user_profile_error_-1) and login with email and password makes the app crashes

### Expected results

Login should be successful

### Project file & Debug log uploads

[debug.zip](https://github.com/user-attachments/files/19743438/debug.zip)

### Checklist of files to include

- [x] Log file
- [ ] Project file

BoBoYu-BamBu, 2025-04-15:
Please provide the DMP log for our analysis. The steps to obtain the logs are as follows：https://wiki.bambulab.com/en/bambu-studio/troubleshoot/crash-freeze-issue

Kindly update it to the latest version V2.0.1.50 on our official website. The download link is：https://bambulab.cn/zh-cn/download/studio

mayl, 2025-09-01:
I still have this issue and am running v2.2.0.85