bambulab/BambuStudio #6525: Camera in my Bambu Lab X1 carbon doesn't work anymore [after an update]. How can I turn it on again? [in the camera settings I can't activate the 'go live' function.

joost-010, 2025-04-15:
### Bambu Studio Version

01.08.05.00

### Where is the application from?

Bambu Lab Official website

### OS version

Windows 10 Pro version 22H2

### Additional system information

Camera in my Bambu Lab X1 carbon doesn't work anymore [after an update]. How can I turn it on again? [in the camera settings I can't activate the 'go live' function.

### Printer

Bambu Lab X1 carbon

### How to reproduce

Camera in my Bambu Lab X1 carbon doesn't work anymore [after an update]. How can I turn it on again? [in the camera settings I can't activate the 'go live' function.

### Actual results

Camera in my Bambu Lab X1 carbon doesn't work anymore [after an update]. How can I turn it on again? [in the camera settings I can't activate the 'go live' function.

### Expected results

Camera in my Bambu Lab X1 carbon doesn't work anymore [after an update]. How can I turn it on again? [in the camera settings I can't activate the 'go live' function.

### Project file & Debug log uploads

Camera in my Bambu Lab X1 carbon doesn't work anymore [after an update]. How can I turn it on again? [in the camera settings I can't activate the 'go live' function.

### Checklist of files to include

- [ ] Log file
- [ ] Project file

Spudhead, 2025-04-15:
My A1 has the same problem. I cant access the camera on my PC or through My ipad.

BoBoYu-BamBu, 2025-04-15:
OK, can we close this list now?

closed 2025-04-28 by BoBoYu-BamBu