bambulab/BambuStudio #6526: Login page from BambuLab Studio 2.0.1.50 not working (blank login form)

chip2me, 2025-04-15:
### Bambu Studio Version

2.0.1.50

### Where is the application from?

Bambu Lab Official website

### OS version

Win 10 Pro

### Printer

A1

### How to reproduce

Start Bambu Studio
Pres login
Wait for login form .... nothing happens ...

Observed also on collegues computer - yeterday and today ...

### Actual results

Login page is not showing anything (white form only)

### Expected results

Expect to have fields for input of username + password

### Project file & Debug log uploads

.

### Checklist of files to include

- [ ] Log file
- [ ] Project file

BoBoYu-BamBu, 2025-04-15:
Thank you for your feedback. Could you try again with a different network?

chip2me, 2025-04-15:
Solved :-) When selecting English in preferences - then it works like expected.
Thanks for suggestion LJKraus.

closed 2025-04-15 by chip2me