bambulab/BambuStudio #6527: No output using CLI in Windows Powershell or Command

arirangz, 2025-04-15:
### Bambu Studio Version

2.0.1.50

### Where is the application from?

Bambu Lab Official website

### OS version

Windonws 10 pro

### Printer

Bambu LAB A1 Mini

### How to reproduce

In terminal when if I run one of the following I don't have any output in the terminal:
.\bambu-studio.exe -h
.\bambu-studio.exe --help
.\bambu-studio.exe "C:\test.stl" --export-3mf "C:\test.3mf" --debug=5

### Actual results

The terminal is no giving any output for help command or debug

### Expected results

As I saw in other issues, I should see output for the help or using the debug.

I also check the AppData log folder but the log is updated only when I launch the GUI application.

### Project file & Debug log uploads

N/A

### Checklist of files to include

- [ ] Log file
- [ ] Project file

arirangz, 2025-04-15:
The only way I found is to redirect to output to a file:
 Start-Process -FilePath "C:\Program Files\Bambu Studio\bambu-studio.exe" -ArgumentList "--help" -NoNewWindow -Wait -RedirectStandardOutput "C:\output.txt"

lanewei120, 2025-04-16:
how about using it under the git bash

arirangz, 2025-04-30:
@lanewei120 
I don't have git bash on this server but I could manage to have output using:

`$pathOut = "C:\output.txt"
 Start-Process -FilePath "C:\Program Files\Bambu Studio\bambu-studio.exe" `
   -ArgumentList '"C:\Program Files\Bambu Studio\test.stl" --export-3mf "C:\test3.3mf" --debug=5' `
   -NoNewWindow `
   -Wait `
   -RedirectStandardOutput $pathOut
`

But would be nice to have the output direclty using bambu-studio.exe in CLI.

lanewei120, 2025-04-30:
yes, it is just output redirectionThis is just an issue of output redirection. We're not quite sure how PowerShell handles it, but it can automatically output on the Linux shell, macOS shell, and Windows Git Bash.

arirangz, 2025-04-30:
@lanewei120 Thanks for your feedback.

So nobody tried in Windows PowerShell or just in standard windows command?

lanewei120, 2025-05-06:
we usually don't try CLI under windows here