bambulab/BambuStudio #6529: Flushing Volumes

Honda16vtec, 2025-04-15:
### Bambu Studio Version

2.0.1.50

### Where is the application from?

Bambu Lab Official website

### OS version

Sequoia 15.3.2

### Additional system information

CPU: Apple M4 Max
Memory: 36GB

### Printer

Bambu Lab X1 Carbon

### How to reproduce

1. Go to flushing volumes

[image]

### Actual results

Flushing volumes pop up window is blank white

### Expected results

Flushing volumes screen is visible and editing volumes is allowed

### Project file & Debug log uploads

It's all projects not just one.

### Checklist of files to include

- [ ] Log file
- [ ] Project file

JohnSmithCYM, 2025-04-16:
We have release V2.0.2(https://github.com/bambulab/BambuStudio/releases/tag/v02.00.02.57) and solve the similar problem with Russian language.

alexjfalcao, 2025-04-17:
I'm still getting the same issue with version 2.0.2.57 (File: Bambu_Studio_mac-v02.00.02.57-20250415200919). Mac M1 Pro Version 15.4. 
[debug_Thu_Apr_17_08_09_11_82884.log](https://github.com/user-attachments/files/19793139/debug_Thu_Apr_17_08_09_11_82884.log)

isrubinger, 2025-04-17:
I also have the same problem here, even with the new version 2.0.2.57, blank flushing screen...

In the previous version it works perfectly.

Waiting for solution

-Bambu Lab P1S
-Mac Mini M1
-Portuguese Language

isrubinger, 2025-04-17:
Update:

After following ticket "Flushing volumes window is empty #6282" it worked, I had 2 installations of Bambu Studio and the new one had a changed name, after deleting the previous version and renaming it only to "Bambu Studio" it worked... Thanks

JohnSmithCYM, 2025-04-17:
Thanks for your feedback, can you turn off the issue now?

Honda16vtec, 2025-04-17:
I still have the same issue even after dowloading the Github version and deleting all previous versions. I also tried changing the name to "Bambu Studio" only and that did not work. 
[image]

alexjfalcao, 2025-04-17:
> I'm still getting the same issue with version 2.0.2.57 (File: Bambu_Studio_mac-v02.00.02.57-20250415200919). Mac M1 Pro Version 15.4. [debug_Thu_Apr_17_08_09_11_82884.log](https://github.com/user-attachments/files/19793139/debug_Thu_Apr_17_08_09_11_82884.log)

I've downloaded again version 2.0.2.57 and installed on a root application folder and now is working fine.
Renamed file or subfolder still not working.

closed 2025-04-17 by Honda16vtec