bambulab/BambuStudio #6531: Ability to add multiple height range modifiers

seanbo, 2025-04-15:
**Is your feature request related to a problem? Please describe.**
No

**Describe the solution you'd like**
I would like the ability to add more than one height range modifier to a project.
**Describe alternatives you've considered**
I cannot think of any options other than just picking which are saves me the most time.

**Additional context**
In my specific scenario, I am printing a Hueforge with a 0.04 layer height.  The first 14-20 layers are usually solid and the top 20 or so layers are as well (used for a border).  At current, I can only modify the layer height to speed up the print time either at the bottom of the print or the top.  Ideally, I would like to be able to do both

[Batman v2.zip](https://github.com/user-attachments/files/19764992/Batman.v2.zip)

JohnSmithCYM, 2025-04-16:
[image]
You can use '+' under ther screen to add multi height modifidier in one object.

seanbo, 2025-04-16:
In my head... I thought that was to adjust the modifier.  so embarrassed.  Thank you.