bambulab/BambuStudio #6532: Storage Tab Issue

jimmystoba, 2025-04-15:
### Bambu Studio Version

2.0.2.57

### Where is the application from?

Bambu Lab Official website

### OS version

Windows 10

### Additional system information

CPU AMD Ryzen 5 4600H
RAM 32GB

### Printer

Bambu P1S

### How to reproduce

With Bambu Studio open
Click on Storage tab
Cannot scroll through Files/timelapses 
eg. if there are more than 9 files stored, I cannot access the older files.
see attached screenshot

[image]

### Actual results

Can only select top 9 files in window.  I have to delete newer files to be able to access older ones.

### Expected results

I think I should be able to scroll through all files stored.

### Project file & Debug log uploads

[log.zip](https://github.com/user-attachments/files/19765744/log.zip)

### Checklist of files to include

- [ ] Log file
- [ ] Project file

bambu123, 2025-04-16:
Are you using a special mouse device? Also, have you encountered this issue in previous versions of Studio?

jimmystoba, 2025-04-16:
Hi, thanks for the swift response.
I'm using a laptop with built in mouse pad.
I also had this problem with the previous version. Only been using Bambu Studio for a few weeks.

JohnSmithCYM, 2025-04-17:
[image]

We are sorry for the fault, but I have tried with my Win11 and everyting is okay. I scroll the build-in mouse pad with two fingers, and it works well.

Can you tell us more about your situation? I am not sure we miss some information here. Can does Studio 1.10 or Orca works well?

jimmystoba, 2025-04-17:
Hi, using two fingers to scroll works. Thank you for your help.
They say every day is a school day. I've had this laptop for about 4 years,
and had no idea you can use two fingers to scroll. 🤣
Thanks, Jimmy.

JohnSmithCYM, 2025-04-17:
Okay, can I turn off the issue now？

jimmystoba, 2025-04-17:
Yes, go ahead.
Thanks again.

closed 2025-04-17 by JohnSmithCYM