bambulab/BambuStudio #6534: Regression in Appearance and Clarity - Checkboxes

FPrefect20, 2025-04-16:
### Bambu Studio Version

2.0.2.57

### Where is the application from?

Bambu Lab Official website

### OS version

Windows 10

### Printer

Bambu X1C

### How to reproduce

**Final Dialog "Send print job"**
Since you mentioned the „easy access” to the 4 option buttons in the dialog "send print job"  in your release notes:
Why worsen a good and easy to use checkbox-control?

"old" Version of the checkboxes:

[image]

"new" version of the checkboxes:

[image]

### Actual results

In my opinion this is a deterioration both in terms of appearance 
(hard to see) and in operation (each control element now has to be clicked twice).

### Expected results

Since the clearly visible checkbox-controls are still used elsewhere in Bambu Studio, please restore them.
Thanks! :-) 

### Project file & Debug log uploads

pls. see screenshots above

### Checklist of files to include

- [ ] Log file
- [ ] Project file

fahaaaxiki, 2025-04-16:
Sorry, due to the recent addition of the "auto" printing option to the printer, Studio now needs to include three optional settings when sending print tasks. As a result, it's no longer possible to use a simple checkbox to control the printing options. We appreciate your support and understanding.

FPrefect20, 2025-04-16:
> Sorry, due to the recent addition of the "auto" printing option to the printer, Studio now needs to include three optional settings when sending print tasks. As a result, it's no longer possible to use a simple checkbox to control the printing options. We appreciate your support and understanding.

?
The number of checkboxes should not be reduced; there should still be 3 (or 4 including "use AMS") checkboxes.
(Please see the screenshots above)