bambulab/BambuStudio #6535: Access violation when slicing

xuehaoying, 2025-04-16:
### Bambu Studio Version

2.0.2.57

### Where is the application from?

Bambu Lab Official website

### OS version

Windows 11 on ARM 

### Additional system information

Device name	BOOK-O2T2OF667S
Processor	Snapdragon(R) X Elite - X1E80100 - Qualcomm(R) Oryon(TM) CPU   4.01 GHz
Installed RAM	16.0 GB (15.6 GB usable)
System type	64-bit operating system, ARM-based processor

### Printer

P1S

### How to reproduce

Slice PETG design with PLA interface layer, set support as Normal instead of tree.

### Actual results

Crash with error message "access violation"

### Expected results

slice completes

### Project file & Debug log uploads

[bambu-studio.exe.30060.zip](https://github.com/user-attachments/files/19767449/bambu-studio.exe.30060.zip)

[test_crash.zip](https://github.com/user-attachments/files/19767451/test_crash.zip)

### Checklist of files to include

- [x] Log file
- [x] Project file

NothFie, 2025-04-16:
@xuehaoying Thanks for your report. This is indeed a BUG that occurs when the "top interface layers" in Normal Support is less than 2. For now you could set it to 2, or use tree support instead and we will fix it in the next version. Sorry for the bad experience you had。

NothFie, 2025-04-30:
The issue is fixed.

closed 2025-04-30 by NothFie