bambulab/BambuStudio #6537: Filament Default View After Slicing

Brudius, 2025-04-16:
### Bambu Studio Version

2.0.2.57

### Where is the application from?

Bambu Lab Official website

### OS version

Windows 11

### Printer

P1P or X1C

### How to reproduce

In the slice preview, it used to remember the Color Scheme set to "Filament" so you can visualize the colors.

### Actual results

 Since the past 2 updates, it only goes to "Line Type".

### Expected results

Please either allow the software to remember the Color Scheme View, or allow you to set a default.

### Project file & Debug log uploads

Not Relevant

### Checklist of files to include

- [ ] Log file
- [ ] Project file

vanowm, 2025-04-16:
It remembers last selected item for me on Windows 10
In v1.9 it always defaulted to `filament` when multiple colors were used - this was very annoying.

Brudius, 2025-04-16:
When it defaults to line, you can't tell if the colors are painted properly and such. I only print with multi-color, so this was a very big quality of life thing. It remembers it in a project, but not from one project to another. It used to always go to the filament, and wish it still would :(.

MackBambu, 2025-04-16:
The default setting for Color Scheme is "Line Type". When you switch to another option, it will remain unchanged unless you restart the software or switch projects. I tested version 1.10 as well, and it behaves the same way. Has there been any change in this behavior?

Brudius, 2025-04-17:
Yeah, In the past, I never had to switch it from "Line Type" to filament type prior to like 2 versions back. I usually operate with 3 different windows open with different projects for my print farm.

closed 2025-06-03 by XunZhangBambu

vanowm, 2025-06-04:
> [@Brudius](https://github.com/Brudius) Hi,This has been fixed in our latest version 02.01.00.59

This issue should remain opened, until the previously selected list is remembered between sessions (I think it should be remembered per project, but default to previous selection if project doesn't have such setting saved).
Current "fix" is just reverted to previous annoying behavior: it defaults to filament list even when single filament was used, which doesn't make any sense.

pcornelius, 2025-06-14:
> [@Brudius](https://github.com/Brudius) Hi,This has been fixed in our latest version 02.01.00.59

Worst fix ever. You have traded one bad behavior for another as it now defaults to filament for the color scheme and still does not remember the last setting used. This might be fine for people who always print with multiple filaments but I, and I think many many others, almost never print with more than one filament as I do not like the waste. So EVERY time I start Bambu Studio and slice a model I have to go and change the Color Scheme to Line Type so I can review the slicing decisions and see if there are problems or areas I can tweak. This is obviously a personal preference which is exactly why Studio should remember the last setting used or allow a preference to be persistently set as to the default color scheme.

Brudius, 2025-06-14:
> Worst fix ever. You have traded one bad behavior for another as it now defaults to filament for the color scheme and still does not remember the last setting used. This might be fine for people who [...]

It literally just went back to how it used to be? I do agree though that there should just be a default view setting where the user can choose which one they want.