bambulab/BambuStudio #6538: Bambu Studio terminates during slicing

hubdows, 2025-04-16:
### Bambu Studio Version

2.0.2.57

### Where is the application from?

Bambu Lab Official website

### OS version

Windows 11

### Additional system information

OS Name	Microsoft Windows 11 Pro
Version	10.0.26100 Build 26100
Other OS Description 	Not Available
OS Manufacturer	Microsoft Corporation
System Name	VE
System Manufacturer	Intel(R) Client Systems
System Model	NUC10i7FNH
System Type	x64-based PC
System SKU	BXNUC10i7FNH
Processor	Intel(R) Core(TM) i7-10710U CPU @ 1.10GHz, 1608 Mhz, 6 Core(s), 12 Logical Processor(s)
BIOS Version/Date	Intel Corp. FNCML357.0063.2023.1024.1117, 10/24/2023
SMBIOS Version	3.3
Embedded Controller Version	3.12
BIOS Mode	UEFI
BaseBoard Manufacturer	Intel Corporation
BaseBoard Product	NUC10i7FNB
BaseBoard Version	K61360-305
Platform Role	Desktop
Secure Boot State	Off
PCR7 Configuration	Elevation Required to View
Windows Directory	C:\WINDOWS
System Directory	C:\WINDOWS\system32
Boot Device	\Device\HarddiskVolume1
Locale	United States
Hardware Abstraction Layer	Version = "10.0.26100.1"
User Name	Not Available
Time Zone	US Mountain Standard Time
Installed Physical Memory (RAM)	64.0 GB
Total Physical Memory	63.8 GB
Available Physical Memory	51.0 GB
Total Virtual Memory	67.8 GB
Available Virtual Memory	54.1 GB
Page File Space	4.00 GB

### Printer

P1S

### How to reproduce

1. Import 3MF
2. Slice
3. CRASH!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!

### Actual results

Bambu Studio crashes. 

### Expected results

PRINT

It was WORKING before the update!!!!!

### Project file & Debug log uploads

[debug_network_Tue_Apr_15_19_09_37.log.zip](https://github.com/user-attachments/files/19769273/debug_network_Tue_Apr_15_19_09_37.log.zip)

[test-v3f.zip](https://github.com/user-attachments/files/19769279/test-v3f.zip)

### Checklist of files to include

- [ ] Log file
- [ ] Project file

hubdows, 2025-04-16:
This bug is preventing my production. I should have never updated!!!

hubdows, 2025-04-16:
What is this? I have NEVER had to enter these values or update them for any of my designs. THIS IS AN ISSSUE!

[image]

hubdows, 2025-04-16:
Fixed model and it still crashes during SLICE.

hubdows, 2025-04-16:
Crashes even if the file is a STL.

hubdows, 2025-04-16:
I went back to the previous version and it's not working now either. Crashing during the slice.

JohnSmithCYM, 2025-04-16:
[image]

The reason why it crashed is due to the "Default filament", change it to any other filament makes slice ok. 
However, can you tell us how this 3mf was made or how you make "Default filament" condition? Thanks for your feedback again.

hubdows, 2025-04-16:
Thanks. I finally caught that at some point last night. However, I had to go back versions since the required firmware 01.08 has not been released yet. The Support site still only has 01.07.

closed 2025-04-16 by hubdows