bambulab/BambuStudio #6540: Can no longer print from spool after latest update 20257

Radicalbradical, 2025-04-16:
### Bambu Studio Version

20257

### Where is the application from?

Bambu Lab Official website

### OS version

windows 11

### Printer

P1s

### How to reproduce

Try to print, and option to disable AMS is gone

### Actual results

unable to print from external spool

### Expected results

should be able to print from external spool or at least have the option available

### Project file & Debug log uploads

this is not file specific

### Checklist of files to include

- [ ] Log file
- [ ] Project file

JohnSmithCYM, 2025-04-16:
We are sorry we did'nt make it clear on release note that we cancel "Use AMS" on X/P/A(like H2D did) and you can select ext spool just click the mapping window to map to ext manually. 

[image]

KelLynnP, 2025-04-16:
[image]

I am not using external pla and cannot change to run the actual filament i am using (PETG).

MartinChan3, 2025-04-16:
That is because the new mapping rule of spool is just like AMS mapping, you can slice in PLA or set external spool filament to PETG to make sure the mapping was right.

XinZhangBBL, 2025-04-23:
@Radicalbradical Sorry, based on our current interaction design, this will not be considered as an option at the moment. But we plan to add some hints to this feature